AprilRobotics/apriltag #263: apriltag3 vs apriltag2 -- detect quad

ruichaowang, 2022-10-17:
**Describe the bug**
Apriltag 3 has a lower detect rate than Apriltag 2.
Tag family:  tagCodes36h11 

**To Reproduce**
Steps to reproduce the behavior:
parameters for apriltag3:
```
    td->nthreads = 8;
    td->quad_decimate = 1;
    td->quad_sigma = 0.0;

    td->qtp.max_nmaxima = 10;
    td->qtp.min_cluster_pixels = 1;

    td->qtp.max_line_fit_mse = 200.0;
    // td->qtp.critical_rad = 1 * M_PI / 180;
    td->qtp.cos_critical_rad = cos(10 * M_PI / 180);
    td->qtp.deglitch = 0;
    td->qtp.min_white_black_diff = 1;

    td->tag_families = zarray_create(sizeof(apriltag_family_t*));

    pthread_mutex_init(&td->mutex, NULL);

    td->tp = timeprofile_create();

    td->refine_edges = 1;
    td->decode_sharpening = 0.25;

    td->debug = 1;
```

**Expected behavior**
Apriltag 3 can detect more tags than Apriltag 2.

** Input Image**
raw imgae: 
[image]

using apriltag2:
[image]

using apriltag3: 
[image]

**Screenshots**
apriltag2 debug, 
[image]

apriltag3 debug, 
[image]

as you can see the apriltag2 can detect more quads.

**Operating Sytem**
m1 mac

**Additional context**
Add any other context about the problem here.

ruichaowang, 2022-10-17:
re-upload raw images.
[image]

christian-rauch, 2022-10-17:
The behaviour between the implementations changed. This repo hosts all the versions. If you need an older implementation, you will have to check out that specific commit.

@mkrogius @wxmerkt Since this also came up in https://github.com/AprilRobotics/apriltag/issues/262, I think it makes sense to create branches `apriltag1` and `apriltag2` for the previous versions 1 and 2. Do you agree? Which commits would match the latest state of the version 1 and 2 implementations?

ruichaowang, 2022-10-28:
> The behaviour between the implementations changed. This repo hosts all the versions. If you need an older implementation, you will have to check out that specific commit.
> It would be great to have branch apriltag2 for the previous versions 2. 
But from my point of view, the apriltag3 should have a better detection rate than apriltag2, but which is not ture in this case.

mkrogius, 2022-11-04:
@ruichaowang Neither AprilTag 3 nor AprilTag 2 supports tags with the extra calibration squares on the corners like that. I also don't recognize the style of the image you have uploaded as `apriltag2_debug`, which makes me wonder whether this is actually an AprilTag 2 implementation? If I had to guess, this would actually be an AprilTag 1 implementation, which didn't officially support tags with calibration squares, but which did have better detection rate in this scenario than AprilTag 3.

@christian-rauch I think that makes sense. Unfortunately none of the earlier AprilTag releases are commits in this repo. It may take some digging in order to find the original releases for AprilTag 1 and AprilTag 2

chjXu, 2023-07-24:
@ruichaowang 哈喽，解决了吗？我也碰到漏检的问题了

ruichaowang, 2023-08-08:
很不幸 没有

christian-rauch, 2024-08-21:
@XavierQL What is confusing? You should rectify the image in any case to have the line detection working properly. Is this related to the AprilTag 2 vs. 3 discussion above?

s-trinh, 2025-04-06:
> Unfortunately none of the earlier AprilTag releases are commits in this repo. It may take some digging in order to find the original releases for AprilTag 1 and AprilTag 2

There is:
- https://april.eecs.umich.edu/media/apriltag/

```
apriltag-0.9.tgz	2015-06-26 20:36
apriltag-0.91.tgz	2015-06-26 20:36
apriltag-0.92.tgz	2015-06-26 20:36
apriltag-2014-10-20.tgz	2015-06-26 20:36
apriltag-2015-03-14.tgz	2015-06-26 20:36
apriltag-2015-03-18.tgz	2015-06-26 20:36
apriltag-2016-10-11.tgz	2016-10-11 23:09
apriltag-2016-10-21.tgz	2016-10-21 20:41
apriltag-2016-12-01.tgz	2016-12-02 01:52
apriltag-2018-11-29.tgz	2018-12-16 17:29
```

I don't know if it is the source code for AprilTag1 but at least there are some previous versions.

s-trinh, 2025-09-22:
For anyone looking, there seems to be a modified Java version of AprilTag1 here:
- https://github.com/icyphy/ptII/blob/master/edu/umich/eecs/april/image/AprilTagFilter.java

````
  <p>The April tags accessor code uses code written by Edwin Olson (ebolson@umich.edu), see
    <a href="https://april.eecs.umich.edu/#in_browser" target="_top">https://april.eecs.umich.edu/</a></p>

  <p>These files are part of a modified subset of the
    april package created by Professor Edward A. Lee (eal@eecs.berkeley.edu) that
    removes all dependencies on external libraries and graphical classes
    and renames the packages to follow Java naming conventions.</p>

  <h2>License</h2>
  <p>Most of April is GPL'd, but we asked for and received a BSD license on the Java april/tag code.</p>
  <p>On 8/20/15 2:38 AM, Edwin Olson wrote:</p>
  <blockquote>
    <p>I am okay with dual-licensing the Java code in april/tag and its necessary dependencies under BSD  clause license. (And since I'm putting this in writing-- this does not extend to other software in the april repository.) The specific license is included below for reference, which includes an additional attribution clause versus the one you sent.</p>

    <p>That said, the java version is so ludicrously outperformed by the C version that I'd really like to convince you to (somehow) use the C version instead. It's not a C-versus-Java thing; the detector logic itself is using completely new algorithms.  It is also *much* cleaner in terms of dependencies.</p>

    <p>Is there some way to make the C version easier to use from Java? I'm no fan of JNI either. Any ideas?</p>
    
    <p>Best,</p>
    
    <p> -Ed</p>
```