AprilRobotics/apriltag #267: Python wrapper on windows not found

parthapn98, 2022-11-28:
Hi Im building april tag using the commands under install here https://github.com/AprilRobotics/apriltag

if i build for -DBUILD_SHARED_LIBS=OFF
The install works properly, howver i need the python wrapper for my work.

Howver when building with -DBUILD_SHARED_LIBS=ON
followed by 
cmake --build build --target install --clean-first
I get an error:
LINK : fatal error LNK1104: cannot open file 'Debug\apriltagd.lib'

and the file itself is missing in the directory.

Any suggestions.
Windows 10, python 3.9.13

christian-rauch, 2022-11-28:
The python wrapper (`BUILD_PYTHON_WRAPPER`) can only be used with shared libraries (see `cmake_dependent_option(BUILD_PYTHON_WRAPPER "Builds Python wrapper" ON BUILD_SHARED_LIBS OFF)`).

At least on Linux, the wrapper is built automatically if `BUILD_SHARED_LIBS` is `ON`. I haven't tried this on Windows. The Python wrapper may just not work on Windows. I can review a PR if someone implements this with a CI pipeline.

parthapn98, 2022-11-28:
is the build tested on unix/ redhat?
it worked fine on Ubuntu

however in windows the build+install without the shared libraries work
but with shared libraries they don't. can you recommend a way for me to share the data with you if you are willing to look into this?

christian-rauch, 2022-12-02:
> is the build tested on unix/ redhat? it worked fine on Ubuntu

I only ever use this on Ubuntu. I assume that this should work on any recent Linux distribution that provides the dependencies. I am not sure if you can compile this on any UNIX variant

> however in windows the build+install without the shared libraries work but with shared libraries they don't. can you recommend a way for me to share the data with you if you are willing to look into this?

Sorry, I don't use Windows and I also do not want to invest the time to set up a Windows development environment in a VM.

The Windows support was generously provided (amongst others) by @ryancwicks (https://github.com/AprilRobotics/apriltag/pull/234) and @VincentMarnier (https://github.com/AprilRobotics/apriltag/pull/240). They may be able to help you.

VincentMarnier, 2022-12-02:
Hello,
I think I might misunderstand something here as I successfuly can build for windows (with and without cross compiling) with both DBUILD_SHARED_LIBS ON and OFF.
I am using the "Developer Command Prompt for VS 2022" (with cmake installed by "Visual Studio Installer").
I run the following commands:
```
cmake -B build -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=ON
cmake --build build --config Release --target apriltag
```
I then have build/Release/apriltag.dll generated.

I tried the python build, installing python on my machine, but ended up on "python_build_flags.py" being called and raising an error. Looking at the file it seems like it just print some Make parameters so it is probably not useful for MSVC anyway.
I never really used python so I doubt I will be of any help if you have to dig this way, sorry.

If this is of any help, here is what I did though:
- I installed python 3.11.0
- I used `pip install numpy`
- I replaced all the `execute_process(COMMAND python3 [...])` by `execute_process(COMMAND python [...])` because for some reason python3 was an issue (even outside of cmake env, using `python3 --version` was raising an error)

I hope you will find your way.
Cheers!

parthapn98, 2022-12-05:
Hi @VincentMarnier 

Thank you for the detailed post.
Basics: i have python3.9 on windows and python3/ python are both bound to it and bypasses the same error you mentioned

the commands i run are 
```bash
cmake -B build -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=ON
cmake --build build --target install
```
note. if target set to apriltag or different name the second command executes successfully, however then the default python wrapper as mentioned in https://github.com/AprilRobotics/apriltag/wiki/AprilTag-User-Guide#python is not installed

the output log for the corresponding commands are as attached
[build.txt](https://github.com/AprilRobotics/apriltag/files/10153363/build.txt)
[install.txt](https://github.com/AprilRobotics/apriltag/files/10153365/install.txt)

The logs are same(allowing for directory naming) if i build withing VStudio

Any idea what could be causing this to happen?

christian-rauch, 2024-01-24:
Fixed via https://github.com/AprilRobotics/apriltag/pull/310.

closed 2024-01-24 by christian-rauch