AprilRobotics/apriltag #272: Border text an issue?

BryanStuurman, 2023-03-03:
I have some apriltags which are being intermittently detected, they have text in the white border (as pictured below) - is this problematic? should the border be clean, and if so for what distance?
[image]

christian-rauch, 2023-03-03:
Yes. The borders have to be clean white so that the edges can be detected. You could try to reduce the size of the text or use a lighter shade of grey. If in doubt, use the [templates](https://github.com/AprilRobotics/apriltag-imgs) or the [tag-generation scripts](https://github.com/AprilRobotics/apriltag-generation).

BryanStuurman, 2023-03-03:
Thanks for the fast response! I'll let my team know.

closed 2023-03-03 by BryanStuurman