AprilRobotics/apriltag #273: Tag bundle transformations

NikHoh, 2023-03-28:
First of all: thank you very much for your awesome work!

From the 'Coordinate system' section in the user guide, it is clear to me how the coordinate systems for the camera and a single tag are defined.

However, could you please provide some information about the transformations in the tags.yaml if I want to define a tag bundle?  

Cheers and best wishes,

Niko

NikHoh, 2023-03-29:
I would like to explain my confusion a little bit further. In the mentioned section of the User Guide, it is said that "The tag's coordinate frame is centered at the center of the tag, with x-axis to the right, y-axis down, and z-axis into the tag."

But from the video on the page [here](http://wiki.ros.org/apriltag_ros/Tutorials/Detection%20in%20a%20video%20stream)

it rather looks like that the frame of a tag is defined (looking at the tag) with the x-axis pointing to the right, the y-axis pointing up and the z-axis pointing out of the tag in direction of us/the camera. 

Is there a mistake in the User Guide or a misunderstanding on my part?

miguelcastillon, 2023-04-28:
Hi!

I just encountered the same problem. I was misled by the wiki, but at least with the dictionary I've used (tagStandard41h12), the reference frame of the bundle seems to be as you describe: "x-axis pointing to the **right**, the y-axis pointing **up** and the z-axis pointing **out of the tag** in direction of us/the camera." The position of the markers in "tag_bundles" needs to follow this convention.

christian-rauch, 2023-06-24:
I am closing this, as tag bundles are a feature of the ROS node. If there is still confusion about the tag bundle frames, you can reopen the issue at https://github.com/AprilRobotics/apriltag_ros.

closed 2023-06-24 by christian-rauch