AprilRobotics/apriltag #284: install problem

Sirhalimi, 2023-07-09:
Hi
I already install all requirements but when I want install in mac and windows get error ( already I had install the libary in another mac )

File "C:\Users\ASUS\AppData\Local\Programs\Python\Python39\lib\site-packages\setuptools\dist.py", line 1234, in run_command 
          super().run_command(command) 
        File "C:\Users\ASUS\AppData\Local\Programs\Python\Python39\lib\site-packages\setuptools\_distutils\dist.py", line 988, in run_command 
          cmd_obj.run() 
        File "C:\Users\ASUS\AppData\Local\Temp\pip-install-fi_9roqm\apriltags_158cb604a2574af99a8b1356cb554008\setup.py", line 35, in run 
          self.build_extension(ext) 
        File "C:\Users\ASUS\AppData\Local\Temp\pip-install-fi_9roqm\apriltags_158cb604a2574af99a8b1356cb554008\setup.py", line 59, in build_extension 
          subprocess.check_call(['cmake', ext.sourcedir] + cmake_args, cwd=self.build_temp, env=env) 
        File "C:\Users\ASUS\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 373, in check_call 
          raise CalledProcessError(retcode, cmd) 
      subprocess.CalledProcessError: Command '['cmake', 'C:\\Users\\ASUS\\AppData\\Local\\Temp\\pip-install-fi_9roqm\\apriltags_158cb604a2574af99a8b1356cb554008', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=C:\\Users\\ASUS\\AppData\\Local\\Temp\\pip-install-fi_9roqm\\apriltags_158cb604a2574af99a8b1356cb554008\\build\\lib.win-amd64-cpython-39', '-DPYTHON_EXECUTABLE=C:\\Users\\ASUS\\AppData\\Local\\Programs\\Python\\Python39\\python.exe', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY_RELEASE=C:\\Users\\ASUS\\AppData\\Local\\Temp\\pip-install-fi_9roqm\\apriltags_158cb604a2574af99a8b1356cb554008\\build\\lib.win-amd64-cpython-39', '-A', 'x64']' returned non-zero exit status 1. 
      [end of output] 
 
  note: This error originates from a subprocess, and is likely not a problem with pip. 
  ERROR: Failed building wheel for apriltags 
  Running setup.py clean for apriltags 
Failed to build apriltags 
ERROR: Could not build wheels for apriltags, which is required to install pyproject.toml-based projects

Sirhalimi, 2023-07-10:
> Does it work in Linux? What happens when you run the cmake command and the arguments manually?

Im not linux user
windows and mac os

christian-rauch, 2023-07-10:
> Im not linux user windows and mac os

That does not really help to solve the problem. Can you provide more details about why the cmake process fails? What commands did you run? Do you actually have CMake installed and is it visible from where you run the script?

Sirhalimi, 2023-07-10:
when I run 'pip install apriltag' this error apear

christian-rauch, 2023-07-10:
> when I run 'pip install apriltag' this error apear

The pip package `apriltag` (https://pypi.org/project/apriltag/) as nothing to do with this repo. Please report these issues to https://github.com/swatbotics/apriltag.