AprilRobotics/apriltag #290: Geometric interpretation of estimation error

afdaniele, 2023-09-09:
Hi, we are trying to use the pose error computed by the function [`estimate_tag_pose`](https://github.com/AprilRobotics/apriltag/blob/6fd978cf03b98d6a058e22e5253cc617c7eccd06/apriltag_pose.c#L528C8-L528C25).

We see values that go from `10e-7` when the detection is qualitatively good to `10e-4` when the detection is in a region of the image that is heavily affected by distortion. What is the unit / geometric interpretation of this error?

Thanks,
- Andrea

mkrogius, 2023-09-17:
This will be in the same units you input for tag size and camera focal length, which usually will be meters. For the geometric interpretation see the paper cited on this line: https://github.com/AprilRobotics/apriltag/blob/6fd978cf03b98d6a058e22e5253cc617c7eccd06/apriltag_pose.h#L50

afdaniele, 2023-10-08:
We will take a look, thanks for the pointer.

closed 2023-10-08 by afdaniele