AprilRobotics/apriltag #296: compatibility with WIN32_LEAN_AND_MEAN

mitjap, 2023-10-27:
This library is not compatible with `WIN32_LEAN_AND_MEAN` definition. It fails to find struct `timeval` which is normally found by `windows.h` header. But with `WIN32_LEAN_AND_MEAN` this header is excluded. 

We could directly include `winsock.h` header as it is specified in official documentation (https://learn.microsoft.com/en-us/windows/win32/api/winsock/ns-winsock-timeval).

https://github.com/AprilRobotics/apriltag/blob/b7bd75ca45d62ac431f10b78b69ea86198cbad14/common/time_util.h#L41

I can prepare a PR for you with this change.

christian-rauch, 2023-10-27:
What's the advantage of using `#define WIN32_LEAN_AND_MEAN`? According to https://learn.microsoft.com/en-us/windows/win32/winprog/using-the-windows-headers:
> Define WIN32_LEAN_AND_MEAN to exclude APIs such as Cryptography, DDE, RPC, Shell, and Windows Sockets.

So it makes sense that some included headers would then be missing.

mitjap, 2023-10-27:
Excluding rarely used headers reduces namespace pollution and improves build time. If one want to use it this library should not stand in the way if possible. Instead of including a very general header it should include exactly what it needs.

christian-rauch, 2023-10-28:
With the diff I better understand what you aim for. I initially understood that you want to use the `windows.h` header with `#define WIN32_LEAN_AND_MEAN` and then additionally add the `winsock.h` header, which would be counterintuitive in my view. Feel free to send a PR with this patch. Also add the required compile flags to the CI in order to make sure that this will not break in future.

Is `timeval` only directly defined in `winsock.h`? Is there no specific "time" header that includes this struct?

mitjap, 2023-10-28:
As specified in documentation `timeval` is defined in **`winsock.h`**. It it also recommended to use `Winsock2.h` which is recommended version of this header.

https://learn.microsoft.com/en-us/windows/win32/api/winsock/ns-winsock-timeval
`Header    winsock.h (include Winsock2.h)`

I am not aware of any other time-related header.

brmarkus, 2023-10-28:
Why is `struct timeval` (still) used at all might be the question... instead of a plain, standard C or C++ way of dealing with "time"?

christian-rauch, 2023-10-28:
> Why is `struct timeval` (still) used at all might be the question... instead of a plain, standard C or C++ way of dealing with "time"?

Do you know of a standard C way of dealing with time at microsecond resolution?

brmarkus, 2023-10-28:
A quick pointer could be "https://stackoverflow.com/a/67731965" (https://stackoverflow.com/questions/5833094/get-a-timestamp-in-c-in-microseconds/67731965#67731965).

Haven't grepped AprilTag for where `timeval` is used... why is a timestamp needed to process AprilTags ;-) ?

christian-rauch, 2023-10-28:
> A quick pointer could be "https://stackoverflow.com/a/67731965" (https://stackoverflow.com/questions/5833094/get-a-timestamp-in-c-in-microseconds/67731965#67731965).

The response you linked recommends not using `timespec_get`. So in the end, you are left with the native APIs. The question and the answer are using the POSIX function `gettimeofday` which is also the one used here.

> Haven't grepped AprilTag for where `timeval` is used... why is a timestamp needed to process AprilTags ;-) ?

The time is used for profiling.

PR #297 (2025-02-16): fix compilation with WIN32_LEAN_AND_MEAN
This PR resolves #296. I changed `windows.h` header for `Winsock2.h`. This way the library still works even when used with `WIN32_LEAN_AND_MEAN` preprocessor definition.

I think even better solution would be to move function `gettimeofday` in `time_util.c` which is the only place where it is used. This way we can also move any required headers (either `windows.h`, `winsock.h` or `Winsock2.h`) from header file which would prevent any side effects to other libraries.

https://github.com/AprilRobotics/apriltag/blob/cc655ad6e3dadb19658276f7476dad5e92f05144 [... cut]
closed 2025-02-16 by christian-rauch