AprilRobotics/apriltag #304: Has anyone confirmed that this library can work on ESP32S3?

ajtwlsdlqp, 2023-11-21:
**Describe the bug**
A clear and concise description of what the bug is.
unknown error occurred

**To Reproduce**
Steps to reproduce the behavior:
[e.g. Run apriltag_demo on the attached input image.]

If you look at the convert.py code,
We are converting *.png files into 8-bit gray images using the cv library and creating data headers.

So, I would like to obtain April-tag information using the data in the frame buffer that stores the ov2640 240 * 240 gray image.

I attached the code as shown in the image below, and when I run it, a core panic occurs.
An infinite reset phenomenon occurs.

[image]

**Expected behavior**
A clear and concise description of what you expected to happen.
run this project at targetboard esp32s3

** Input Image**
If applicable, please attach the input image that reproduces the problem.
[image]

**Operating Sytem**
[e.g. Ubuntu, iOS, Windows]
Target Board ESP32S3 N8R8
ESP-IDF 4.4.6

** Installation Method**
[e.g. I built AprilTag from source following the instructions in the README]
follow both method
https://components.espressif.com/components/liux-pro/apriltag
https://github.com/liux-pro/apriltag

**Additional context**
Add any other context about the problem here.

Here's a look at some of the errors I've made in the past.
https://github.com/liux-pro/apriltag/issues/1

Need help 
Best Regards.

christian-rauch, 2023-11-21:
I assume that only very few people will have access to such a specific board. Is this reproducible on more widely available hardware such as a "regular" x86 PC or a Raspberry Pi? Can you provide a detailed backtrace of where the crash happened and how the functions of the AprilTag library are involved in this?

ajtwlsdlqp, 2023-11-22:
@christian-rauch 
Thanks for help.

the other DL library such as face-recognition, move detection, color recognition, web streaming are working well.
and also, if i change mode my board to work for face detect it also work well, however when i change mode to detect april-tag esp32s3 board reset directly.

in normally, esp terminal show up the error reason such as Buffer Overflow or else, but in this time any message are not come out. 

that's why i assume that

christian-rauch, 2024-01-21:
Unless this can be reproduced on standard hardware, an emulator, etc., I don't see how we will be able to determine if this issue lies within the library, and if so, fix the issue. I am going to close this for now. Please reopen, if you find a way to reproduce the issue.

closed 2024-01-21 by christian-rauch