AprilRobotics/apriltag #306: How to use intrinsics in Scaramuzza format to estimate the apriltag pose?

Ttester007, 2023-12-29:
Hi,

Instead of camera matrix K and distortion coefficient vector D, I have the camera intrinsics in Scaramuzza format (polynomial coefficients, Inverse Polynomial coefficients, affine coefficients and Principal point). Is there any way to use these intrinsics to detect the april tag and estimate the pose accurately? 

For your reference, Scaramuzza Model details can be found here : [link](https://sites.google.com/site/scarabotix/ocamcalib-omnidirectional-camera-calibration-toolbox-for-matlab)  

Thanks

christian-rauch, 2023-12-29:
Is this specifically about omnidirectional cameras? I have no experience using omnidirectional cameras so I won't be able to answer all your questions. AFAIK, the apriltag library does not support omnidirectional cameras directly out of the box.

The `apriltag` library expects rectified images to detect the tags. At least, you have to rectify the images and provide them to the library to detect the tags. To estimate the pose, you will need the camera parameters, which is currently only implemented for the pinhole camera model.

Ttester007, 2023-12-29:
Thanks for the reply. 
Is there any way we can do modification to april tag detector library so that we can use Scaramuzza format intrinsics?

closed 2024-05-09 by christian-rauch