AprilRobotics/apriltag #321: What's the meaning of 'apriltag.c:448: homography_compute2: Assertion `max_val_idx >= 0' failed.'?

WZG3661, 2024-03-23:
an error ccurred accidentally:
apriltag.c:448: homography_compute2: Assertion `max_val_idx >= 0' failed.
what's the meaning of this error? did I do something wrong?

christian-rauch, 2024-03-23:
This check was added in #317 to avoid indexing with a negative index. I hope that the proposed fix in #213, which was ported into #317 via 359a98403e64237d6ec7b800dc8b61ddaa7ee570, fixes the issue, but apparently, it does not.

Do you have a reproducible script and data for this?

WZG3661, 2024-04-03:
this is my code:
```C++
#include "apriltag.h"
#include "tagStandard41h12.h"

#include <iostream>
#include <opencv2/core/core.hpp>
#include <opencv2/highgui/highgui.hpp>
#include <opencv2/opencv.hpp>

int main() {

  std::string img_path = "/home/lymow/Test/datasets/lymow/2024-03-14-10-13-56/right_camera/1710383687.256000.jpg";
  cv::Mat img = cv::imread(img_path, CV_8UC1);

  apriltag_family_t *tf;
  apriltag_detector_t *td;

  tf = tagStandard41h12_create();
  td = apriltag_detector_create();
  apriltag_detector_add_family(td, tf);

  image_u8_t im = {img.cols, img.rows, img.cols, img.data};
  zarray_t *detections = apriltag_detector_detect(td, &im);

  apriltag_detector_destroy(td);
  tagStandard41h12_destroy(tf);
  return 0;
}
```
this is the picture:
[image]

mkrogius, 2024-05-05:
At e6cd3c7afdc23814f38a6b537904a41c393fc305 I am able to repro with:
```
rm -r build
cmake -B build
cmake --build build
uild/apriltag_demo -f tagStandard41h12 ~/Downloads/319063435-0f7a44f8-58e7-452c-99bc-142160e24a93.jpg
```
(I wasn't able to repro with -DCMAKE_BUILD_TYPE=Release since it removes asserts)

Printing out the correspondences matrix that is input to homography_compute2 in this case gives:
```
-1.0000 -1.0000 191.9706 454.0736 
1.0000 -1.0000 191.9706 454.0736 
1.0000 1.0000 191.9706 472.9767 
-1.0000 1.0000 191.9706 472.9767 
```
Which is a degenerate case for which a valid homography does not exist. **We should return null rather than asserting from homography_compute2.**

But how did we end up with a degenerate case not getting filtered out earlier? This is the quad in question:
[image]

The reason is that we only end up with a degenerate quad after the refine_edges step and there are not checks on the result of the refine edges step that the quad is valid. This is fine IMO since the fastest way to know if a quad is valid at this point is just to do the decode step, which is what happens next

PR #325 (2024-05-06): fix homography assert
When the "Find best row to swap with." section in `homography_compute2` would not find a value greater than 0 in a row, `max_val_idx` would stay `-1` and maybe be used later to index a matrix. We protected against the negative indexing (in debug mode) by checking for a positive index in 3a0a155d5bc3bdcef34275f7c51f56e9cfd34e13.

But this causes issues for false-positive quad detections (#321). In this particular case, ignoreing the negative index would lead to `return NULL;` via `max_val < epsilon`, so we can also just `return NULL;` here and ignore that quad instead of s [... cut]
closed 2024-05-06 by christian-rauch