AprilRobotics/apriltag #322: Planned Python Package Index (PyPI) package?

Jgunde, 2024-03-25:
There are lots of other AprilTag packages on PyPi that can be installed easily with pip. But none of them seem as robust and well maintained as this one. Is it possible to install this library with pip?

christian-rauch, 2024-03-25:
Not at this moment. Someone would need to create a Python project description (`pyproject.toml`) or similar to generate a wheel file from the source.

Jgunde, 2024-03-25:
Ok thank you!

closed 2024-03-25 by Jgunde

sgbaird, 2024-05-15:
@christian-rauch if I were to take a shot at this, is there a possibility of getting this integrated into the main project?

christian-rauch, 2024-05-15:
Sure. If you add a CMake target that generates the wheel file or similar, you can send a PR for review.

If you need a starting point for this, I suggest https://github.com/stevenlovegrove/Pangolin as it has a CMake target that builds the wheel file (`cmake --build build -t pypangolin_wheel`).