AprilRobotics/apriltag #326: Build fail on Ubuntu Noble in g2d.c: `last_quadrant` is used unitialized; treat-all-warnings-as-errors

ojura, 2024-05-06:
Build fails because of unitialized variable use:

```
  /<<BUILDDIR>>package/common/g2d.c: In function ‘g2d_polygon_contains_point’:
  /<<BUILDDIR>>package/common/g2d.c:351:17: error: ‘last_quadrant’ may be used uninitialized [-Werror=maybe-uninitialized]
    351 |             int dquadrant = quadrant - last_quadrant;
        |                 ^~~~~~~~~
  /<<BUILDDIR>>package/common/g2d.c:324:9: note: ‘last_quadrant’ was declared here
    324 |     int last_quadrant;
        |         ^~~~~~~~~~~~~
```

christian-rauch, 2024-05-06:
This should be fixed by https://github.com/AprilRobotics/apriltag/pull/324. I could not reproduce the issue with the 24.04 Docker image. Can you check if the PR solves your issue?

christian-rauch, 2024-05-07:
Can you provide instructions on how to reproduce this? Which CMake generator and compiler are you using? Is this in a Docker container or a "bare metal" installation?

closed 2024-05-09 by christian-rauch

ojura, 2024-05-09:
Hi Christian, sorry for missing your response. I am glad that you figured it out and thanks for the fix. For me, it was clear from the source code that the `last_quadrant` variable was uninitialized, and I didn't consider it necessary to include more detailed reproduction info.

christian-rauch, 2024-05-09:
I could see that it was not initialised too. But I was still curious how to reproduce the error locally as I have seen this issue on other CI pipelines too but I could reproduce this in a Docker container. Anyway, this should now be fixed.