AprilRobotics/apriltag #328: How to detect  AprilGrid?

wfxcqsc, 2024-05-10:
hello,I am confused with the AprilGrid, when i detect the following picture（Apriltag board） using the opencv_demo, it can be successfully detected.
[image]

however,when detect the AprilGrid, it can not detect, i guess the reason is the the border width of Apriltag board is one ，while AprilGrid with border width the size of two
[image]

so，i want to figure out whether can set the borderbits, and how to detect AprilGrid?

Looking forward to your reply!

christian-rauch, 2024-05-10:
The apriltag library needs a white border around the tag to do the edge detection properly. This border is interrupted by the black squares on the corner of the tags.

wfxcqsc, 2024-05-11:
thanks for your reply!
however, i have tried to cover up the black squares in the AprilGrid，and detect again. but it still doesn't work.
[image]

then,  i detect the same picture get from the above using OpenCV aruco , there is a"DICT_APRILTAG_36h11", if i don't set the "markBorderBits"(default value is 1), it also can't detect. while i set the parameter "markBorderBits=2", it can be successfully detect 

[image]
Therefore, in addition to black square interruption, is the border width also a part of the impact on detection? and whether the apriltag library can detect the APRILGRID?

wfxcqsc, 2024-05-11:
thanks for your reply!
i am sorry about my issue，i put the wrong image in the first picture. the Apriltag board that can be Apriltag library successfully detect is the following
[image]

however, i have tried to cover up the black squares in the AprilGrid，and detect again. but it still doesn't work.
[image]

then, i detect the same picture get from the above using OpenCV aruco , there is a"DICT_APRILTAG_36h11", if i don't set the "markBorderBits"(default value is 1), it also can't detect. while i set the parameter "markBorderBits=2", it can be successfully detect
[image]

Therefore, in addition to black square interruption, is the border width also a part of the impact on detection? and whether the apriltag library can detect the APRILGRID?

mkrogius, 2024-05-11:
The apriltag library only supports a black border width of 1. It looks like in your image the border width is 2, which is why it is not detected

christian-rauch, 2024-05-27:
@wfxcqsc I tried your AprilGrid image with `apriltag_demo` and `opencv_demo` and I got a couple of detections. Do you not get any detections?

Alternatively, you can also try the "AprilTags C++ Library" from https://people.csail.mit.edu/kaess/apriltags/ (https://bitbucket.org/kaess/apriltags/src/master/) which generally supports tag families with multiple black border sizes: https://bitbucket.org/kaess/apriltags/src/3aea96d3239bb83870cb5e0a85f142a2b0589563/AprilTags/TagFamily.h#lines-64.

wfxcqsc, 2024-05-28:
Thanks for your reply！in fact, when I using opencv_demo to detect AprilGrid which have black squares on the corner of tags, I can not detect any detections. but I have detected AprilGrid successfully by using Kalibr，and I will also try the “AprilTags C++ Library”.

closed 2024-05-28 by wfxcqsc