AprilRobotics/apriltag #333: Question: Use more than 4 points for pose computation

s-trinh, 2024-06-05:
Sorry, this is not a bug but rather a question.

---

I am wondering if using not only the 4 corners of the marker but some specific (depending on the marker ID) additional corners from the pattern itself would bring more accurate pose estimation?
For instance, when processing time is not a constraint, a first estimation of the marker pose could be computed as usual and then refined ([`orthogonal_iteration()`](https://github.com/AprilRobotics/apriltag/blob/c4449c7bca8a6e4cc7f032bf06869f6a81e5ce74/apriltag_pose.c#L43)?) using some additional points.

[image]

I would say maybe in theory this should bring more accurate pose estimation, but in practice this should be useless wrt sensor noise, image distortion, etc.
I think that the 4 corners coordinates are already computed from contour points line fitting, so using additional corners from the pattern should not bring additional useful information?

I am interested about other opinions.

mkrogius, 2024-06-05:
I tried something like that but it didn't seem to give improvement, at least for far away tags. My guess is that the edge fitting already makes use of most of the useful information for pose.

Some previous discussion here: https://github.com/AprilRobotics/apriltag/issues/19

s-trinh, 2024-06-06:
Thanks for your feedback.
Maybe you could add a specific label for questions or enable [Github discussion](https://docs.github.com/en/discussions/quickstart) tab?

kke19, 2024-06-25:
okay, thank you