AprilRobotics/apriltag #335: ReadMe File Error

Suave101, 2024-06-07:
On the ReadMe file of the project here:
[image]
Perma-Linked: https://github.com/AprilRobotics/apriltag?tab=readme-ov-file#python
The link that I boxed in red is incorrect. It directs you to a repo that is no longer supported. Here is the correct link:
https://github.com/duckietown/lib-dt-apriltags
Here is a picture of the unsupported repo that the link currently links to:
[image]

Suave101, 2024-06-07:
Here is a pull request I made to fix the issue:
#336

christian-rauch, 2024-06-09:
Fixed via https://github.com/AprilRobotics/apriltag/pull/336.

closed 2024-06-09 by christian-rauch