AprilRobotics/apriltag #342: from apriltag import apriltag no module

ehswnmohseni, 2024-07-24:
hello i encountering an error when importing the apriltag module after installing it using the provided guide from https://github.com/AprilRobotics/apriltag?tab=readme-ov-file#install. The error message ModuleNotFoundError: No module named 'apriltag' indicates that Python cannot find the installed apriltag module.

how can i solve this error ?

i`m beginner.

ehswnmohseni, 2024-07-27:
I created a virtual environment to resolve this issue and placed the 'apriltag.cpython-36m-x86_64-linux-gnu.so' file into the 'venv/lib/python3.12/site-packages/' directory. This successfully resolved the problem.

closed 2024-07-27 by ehswnmohseni

christian-rauch, 2024-08-02:
The `apriltag.cpython-312-x86_64-linux-gnu.so` (or similar on other systems) has to be in the search path for Python modules. With:
```sh
cmake -B build -GNinja -DCMAKE_BUILD_TYPE=Release
cmake --build build
```
this file will be located on the `build` folder. Running the Python interpreter from this build folder will make the module available:
```sh
cd build
python3 -c "from apriltag import apriltag; detector = apriltag('tagStandard41h12')"
```

Of course, if you want to access the module from anywhere, you have to move it to one of the search paths. You can directly install to the user location via `cmake --build build --target install` as per documentation.