AprilRobotics/apriltag #343: quick_decode struct allocates  6Gb memory for  standard52h13 family

kde-baskets-user, 2024-08-13:
**Describe the bug**
quick_decode_init() allocates too much memory for  the standard52h13 family.  Any workarounds?

**To Reproduce**
Create standard52h13 detector to see allocated memory.

**Expected behavior**
Parsing of 15 bits of useful data  from an image should take more reasonable amount of memory. 

**Operating Sytem**
Ubuntu

** Installation Method**
I built AprilTag from source following the instructions in the README

** Code version **
3.4.2 from GitHub

christian-rauch, 2024-08-16:
I can confirm the high memory consumption by running `./opencv_demo --camera 2 --family tagStandard52h13` and checking the memory consumption with `top -p $(pidof opencv_demo)`, which reports around 6GB RES consumption.

The `tagStandard52h13` has one of the largest code books (48714 codes). Loading an even larger code book from the `tagCircle49h12` (65535 codes) will consume even more memory (more than 7GB in my case).

> Parsing of 15 bits of useful data from an image should take more reasonable amount of memory.

How do you get to 15 bits? As the name `tagStandard52h13` suggests, this tag family encodes 52 bits with a hamming distance of 13.

If you need to reduce the size of the code book, I simply suggest using a different tag family with fewer encoded bits. You can also [generate](https://github.com/AprilRobotics/apriltag-generation) your own tag family for specific requirements.

kde-baskets-user, 2024-08-17:
Thanks for your answer.  Since tagStandard52h13 has 48714 different entries, one aprilag code contains 15.6 = log2(48724) bits of information, by "useful data" I meant rigorously defined word "information", sorry for any confusion.  Everything else in 52  bits representation has to do with code correction, I assume.  

Now, while memory depending on ncodes grows only polynomially(and not even quadratically, as far as I see), 6Gb for many embedded applications is prohibitively costly especially for a seemingly modest task of error correction, but I am not expert in this field. 

I am currently evaluating several 2D barcode libraries and this is so far a show-stopper for apriltag. I am going to run some more tests and perhaps will be back with more data on this.

christian-rauch, 2024-08-17:
> Thanks for your answer. Since tagStandard52h13 has 48714 different entries, one aprilag code contains 15.6 = log2(48724) bits of information, by "useful data" I meant rigorously defined word "information", sorry for any confusion. Everything else in 52 bits representation has to do with error correction, I assume.

Yes, most of the bits are for the hamming distance, e.g. to allow a certain amount of bit errors before codes / IDs are mixed up. If you need many IDs with a large hamming distance, then you just have to use many bits. If you do not need to have codes with at least a certain hamming distance, you can just use a 16bit tag family.

> Now, while memory depending on ncodes grows only polynomially(and not even quadratically, as far as I see), 6Gb for many embedded applications is prohibitively costly especially for a seemingly [...]

I would be curious to see alternative solutions to this.

If I may, why are you trying to distinguish that many unique IDs in an embedded application? If memory is a concern, why are you not using a smaller tag family? E.g. the `tag36h11` has more than 500 unique IDs with a minimum hamming distance of 11 and consumes far less memory.

closed 2024-08-28 by christian-rauch

mkrogius, 2025-08-09:
In the "Choosing a Tag Family" section of the readme we do say "For the vast majority of applications, the tagStandard41h12 family will be the correct choice.". It is true that this section could be improved to elaborate on why exactly other tag families might not be the right choice. But also it is reasonable for us, with the limited volunteer time available, to expect that users will self serve if straying from the beaten path. In this particularly case there is the very simple solution suggested by @christian-rauch above of setting `maxhamming` to a lower value, which will greatly reduce the memory required

nathanieltagg, 2025-08-09:
I suggest updating the documentation: "The tagStandard52h13 family offers a larger range encoded values, but at the expense of very large (6GB) memory use in the tag detector."  And in the discussion of options: "choosing a maxhamming value of xxx will result in a much lower memory requirement, which can be important in when using the 52h13 family".

Without this information, I chose the 52 family thinking it would give me the option of a nice large semantic numbering system,  maybe even encoding a few bytes of text.  I didn't need this, but the downside was -never mentioned-; I assumed that the costs were minor.  

This wasn't the case, and I spent 2 days debugging a massive ram impact on our commercial robots.

christian-rauch, 2025-08-09:
Those are oddly specific. Even larger tags will require even larger amount of memory. I don't think providing this specific information for every tag family is useful. But if you find it useful, there could be more information in the "Choosing a Tag Family" about how to select a tag family or what to look out for when generating your own tag family.

If you are using one of the predefined tag families from the repo, I guess you should first decide how many different tags you have to distinguish and then select a hamming distance required for your environment. If you only need to distinguish 100 tags, there is no point in selecting a tag family providing 48714 unique IDs.

How many different IDs do you actually need?

> Without this information, I chose the 52 family thinking it would give me the option of a nice large semantic numbering system, maybe even encoding a few bytes of text.

At 52 bit with "only" 48714 codes, most of the encoded integer values will be invalid and used for the hamming distance. You won't be able to encode arbitrary strings of text this way.

> I didn't need this, but the downside was -never mentioned-; I assumed that the costs were minor.

If you did not need that many unique IDs, why did you start with the largest pre-define tag family instead of starting with a smaller one? Generally, for every engineering problem, I would recommend to start with the simplest solution and then increase the complexity until the problem is solved. In your case, I would start with the tag family that has at least the amount of codes (see `ncodes` in the source file) that you require for your application, and then test with example data how much hamming distance you need at minimum for a robust detection.

I guess these recommendations could be part of "Choosing a Tag Family". If you find this helpful, I would be happy to review a PR with updates to the documentation for selecting an appropriate `ncodes` vs. hamming distance.

nathanieltagg, 2025-08-17:
> If you did not need that many unique IDs, why did you start with the largest pre-define tag family instead of starting with a smaller one? 

Simple: there seemed to be no downside!  **Because there was no discussion of cost**, I assumed that it simply meant the tags were harder to resolve if small, which wasn't an issue.  It meant I didn't have to ever worry about collisions, which was convenient but not a requirement.