AprilRobotics/apriltag #347: coredump when apriltag detection

biubiubiu554, 2024-09-03:
**Describe the bug**
coredump occured when some pictures was used to detect apriltag. the stack information  as follows:

0# std::basic_ostream<char, std::char_traits<char> >& std::operator<< <std::char_traits<char> >(std::basic_ostream<char, std::char_traits<char> >&, char const*) at /usr/bin/../lib/gcc/x86_64-linux-gnu/7.5.0/../../../../include/c++/7.5.0/ostream:561
 1# 0x00007F1BECE48F10 in /lib/x86_64-linux-gnu/libc.so.6
 2# pthread_mutex_lock in /lib/x86_64-linux-gnu/libpthread.so.0
 3# 0x00007F1B25C18FA0 in /usr/local/libtest.so
 4# workerpool_run(workerpool*) in /usr/local/[libtest.so]
 5# apriltag_detector_detect(apriltag_detector*, image_u8*) in /usr/local/libtest.so]

by  debug the program，the error occurs in function :
void fit_line(struct line_fit_pt *lfps, int sz, int i0, int i1, double *lineparm, double *err, double *mse)

when the line length is 0, the line param is nan, the coredump will occured in the next matrix copute.

i can fix the problem by nan judgement, but i do not know why the line length would be zero and why the dump in the pthread_mutex_lock .

christian-rauch, 2024-09-03:
Can you provide a reproducible example (preferably via `apriltag_demo`) and the original image that you used?

christian-rauch, 2024-09-04:
> I use the old version (V3.1.5)

Is there a particular reason why you are using such an old version? The latest release is `v3.4.2`.

I cannot reproduce a crash with either version. When I run your code with `v3.1.5`, I get this result:
[image]
and on the terminal, I see:
```
image channels: 3
QSocketNotifier: Can only be used with threads started with QThread
WRN: Matrix is singular.
WRN: Matrix is singular.
tag size: 3
```

> I test the latest version will not coredump.

I assume that this is related to the `WRN: Matrix is singular.` issues that have been fixed recently. If the newer releases with these fixes work for you, I strongly suggest switching to a newer version.

biubiubiu554, 2024-09-05:
The problem has been resolved.

closed 2024-09-05 by biubiubiu554