AprilRobotics/apriltag #350: pose estimation issue about Tags without black borders

zacario-li, 2024-09-12:
When I try to estimate the pose with 'Standard41h12,' 'Standard52h13,' 'tagCustom48h12,' etc., which do not have black borders, I believe the 'tagsize' should be modified to represent the inner corner edge length rather than the outer edge length.(Refer to the code [here](https://github.com/AprilRobotics/apriltag/blob/620b84234dbe61cb8f1aae64beed34c8258004c1/apriltag_pose.c#L498)). 
I tested using the inner edge length, and the translation accuracy is improved compared to using the outer edge.

christian-rauch, 2024-09-12:
Isn't the tag length already defined as the inner length? See https://github.com/AprilRobotics/apriltag?tab=readme-ov-file#pose-estimation which specifically states:

> Note: The tag size should not be measured from the outside of the tag. The tag size is defined as the distance between the detection corners, or alternately, the length of the edge between the white border and the black border.

and then shows the corner points of the edges.

zacario-li, 2024-09-13:
Oh, sorry,  thanks for your infomation.

closed 2024-09-13 by christian-rauch