AprilRobotics/apriltag #355: Porting to single core embedded

Chris-F5, 2024-09-26:
For my application, I'd like to detect april tags on a single core of an RP2040. This means there is no `<pthread.h>` for me to include and use. I'm happy to try and implement this port myself, but I create this issue for 3 reasons:

1. Has anybody attempted this before? Is there some existing port I haven't been able to find?
2. Because I'm open to input on how the best way to implement this would be.
3. Is this something that would be usefull to have in the main repo. I.e. should I make a pull request once I'm done?

At first glance, it seems that a good way to do this would be to add a new macro option in `common/pthreads_cross.{c,h}`, similar to `_WIN32`. Something like:
```
#ifdef _NOTHREADS
typedef void pthread_mutex_p;
...
```
And then add some checks that only one thread is being created elsewhere. By doing this I would be eliminating all dependence on `<pthread.h>` when `_NOTHREADS` is set.

christian-rauch, 2024-09-26:
I would be up for making threading optional via a CMake option. If threading is deactivated, then everything should behave via `#ifdef`s as if a single thread was created.

Chris-F5, 2024-09-30:
[Here](https://github.com/Chris-F5/apriltag/commit/0ce68e62af5e14d86b3c8e63133c6f395dd2b05c) is my NOTHREADS macro implementation. If `NOTHREADS` is defined then `pthreads_cross.{c,h}` will declare and define `pthread_*` functions which return immediately (this is just so that all symbols will get resolved). `workerpool.c` will then enforce a single thread. I'm not so great with CMake so I haven't added an option in that yet.

Some other things I noticed when trying to make my rp2040 micropython port:

- `pthreads_cross.{c.h}` were using CRLF line ends. (pull request created)
- `int32_t` types are being foratted with `%d` which assumed that `int` is 32 bits. `int32_t` types should be formatted with `PRId32`. See the [docs](https://en.cppreference.com/w/c/types/integer). I have fixed some of these in [this commit](https://github.com/Chris-F5/apriltag/commit/e6ce29b3dfec6ba5033101ee2f90e3fbf201cf72) but its possible I missed some, since I'm not using the full source tree. I can create a pull request if you want.

If you need to detect apriltags on an embedded device running micropython, you can check out [the branch on my fork](https://github.com/Chris-F5/apriltag/tree/micropython).

christian-rauch, 2024-09-30:
For the "single-thread at compiletime" feature, send a PR so it can be reviewed properly. Disabling the threading and the workpool would generally be useful.

For the format specifier, it would be useful to enable warnings or errors so that this issue does not reappear when a new `fprintf` is added.

christian-rauch, 2025-02-16:
Closing this as there does not interest and support of compiling the library without threading at the moment (https://github.com/AprilRobotics/apriltag/pull/359#issuecomment-2661381681).

closed 2025-02-16 by christian-rauch