AprilRobotics/apriltag #365: Hey there, I would like to request a feature in apriltag.c  which allows to add a ROI of which the indexes of the iterators that scan tags in the first stage of the detect method would be marginalized to reduce processing time

GGreenix, 2024-12-24:
I'm currently working on the same principle but encountered I need to recalculate the homography, if we would ROI the indexes directly the detection would still be in the original coordinates but faster wouldn't it? Thanks in advance :).

GGreenix, 2024-12-24:
Btw if you have a solution for it feel free to suggest.

christian-rauch, 2024-12-24:
If I understand you correctly, you just want to provide a cropped region of an image to the detector. You can do this already. Just crop the image, recompute the intrinsics, and then provide this instead of the full image and intrinsics to the detector.

Are you requesting this recalculation of the intrinsics as a feature of the apriltag library?

GGreenix, 2024-12-24:
> If I understand you correctly, you just want to provide a cropped region of an image to the detector. You can do this already. Just crop the image, recompute the intrinsics, and then provide this [...]

Yes but we discovered that you need to recalculate the homography in order to get accurate results for 3D localization, we offset now the cx, cy elements by the offset of the new x,y but maybe there's something else that effects the homography? We thought it could be solved by directly only scanning for tags in a ROI with the original coordinates

GGreenix, 2024-12-24:
Btw what do you mean by intrinsically

GGreenix, 2024-12-25:
> If I understand you correctly, you just want to provide a cropped region of an image to the detector. You can do this already. Just crop the image, recompute the intrinsics, and then provide this [...]

excuse me for the last message, how can we recalculate the camera intrinsics? could you point me to the method so we could use it and what are its parameters?

christian-rauch, 2024-12-28:
> excuse me for the last message, how can we recalculate the camera intrinsics? could you point me to the method so we could use it and what are its parameters?

I am not going into details as this is a generic feature for 2D-3D projection and I wouldn't add something like this to the apriltag library.

When you crop the image such that the field-of-view (FoV) changes, you have to adjust the focal length accordingly. If you shift the cropped region, the centre point will shift too accordingly. Have a look at the documentation of the "pinhole camera model" in the OpenCV documentation for example: https://docs.opencv.org/3.4/d9/d0c/group__calib3d.html.

closed 2024-12-28 by christian-rauch