AprilRobotics/apriltag #367: Recalculating intrinsics

GGreenix, 2025-01-04:
I am sorry for repeating this, this is just a question now, I tried achieving the recalculation of the tomography after offsetting my quad corners after image cropping, but it seems to have come off, what is the correct way to recalculate the intrinsic?
could you point me to the correct methods?
also in what form are the quad corners, I saw it is CCW but what corner is the beginning?

again sorry for repeating and thank you for replying, have a good week.

GGreenix, 2025-01-05:
I want to note that I copied the method of quad_update_homography,
At line 488 at apriltag.c, and maybe I didn't give it the correct parameters but it didn't seems like it gave similar answers when not cropping.

christian-rauch, 2025-01-14:
> I am sorry for repeating this, this is just a question now, I tried achieving the recalculation of the homography after offsetting my quad corners after image cropping, but it seems to have come off, what is the correct way to recalculate the intrinsic?

If you just want to transform coordinates between the 2D coordinate systems of the original image and the cropped ROI area, you just need to add the 2D offset of the ROI inside the image to the extracted coordinates.

> also in what form are the quad corners, I saw it is CCW but what corner is the beginning?

You could just test this yourself by inspecting how the debug images are drawn:
https://github.com/AprilRobotics/apriltag/blob/fc2a7b20a49fc8b211709280e337a76a30db3042/apriltag.c#L1134-L1156

If you look at the tag, the first corner starts in the top right and then goes clockwise from there, i.e. (`p[0][0]`, `p[0][1]`) is top right, (`p[1][0]`, `p[1][1]`) is bottom right, (`p[2][0]`, `p[2][1]`) is bottom left, etc., such that the first `image_u8_draw_line` draws the right edge, the second `image_u8_draw_line` draws the bottom edge, and so on.

I am closing this now since your issues are not related to the apriltag library. If you think there is something important missing from the documentation, the open an issue with the concrete statements of what information is missing.

closed 2025-01-14 by christian-rauch

GGreenix, 2025-01-14:
And what about the recalculation of the homography? Is there a way to do
it? I copied the method called compute_homography2() but seems like it
wasn’t enough

GGreenix, 2025-01-14:
A follow up question to your answer about the 2D coordinates, I cant add it
to the quad in the C code because it causes weird issues I've tried that,
where can I add the Offset exactly?
I tried looking for the specific piece of code which is before the
homography calculation but didn't find any that didn't crash, I am not as
profound in the algorithm so modifying it is very hard for me, in my code
itself I do add the offset but still needs to recalculate the homography so
I still need a solution for that, thank you for answering

‫בתאריך יום ג׳, 14 בינו׳ 2025 ב-22:58 מאת ‪Christian Rauch‬‏ <‪
***@***.***‬‏>:‬

christian-rauch, 2025-01-14:
> And what about the recalculation of the homography? Is there a way to do it?

As you can see from the code, `quad_update_homographies` is computing the homography from the `quad` corners. So you should just need to translate your quad corners from the detections in your cropped ROI area back to the image and continue with the rest of the pipeline, including the `quad_update_homographies`.

> I cant add it to the quad in the C code because it causes weird issues I've tried that, where can I add the Offset exactly? I tried looking for the specific piece of code which is before the homography calculation but didn't find any that didn't crash,

I am not going to solve this problem with your code unless you can show that this is caused by the apriltag library itself and not your code. I don't see why it would not be possible to modify the values in `quad`.

---

Please stop creating issues in this repo to ask questions about the usage or generic topics like camera intrinsics. I don't think that what you want is going to be implemented in this repo. You will have to do it yourself or ask for help in a forum or something like https://robotics.stackexchange.com.

The issues are meant to report bugs or other unexpected behaviour with the code. You can also use the PRs to fix bugs yourself or propose useful features. But IMHO, recalculating the intrinsics for an ROI is too generic and can be done outside of the apriltag library. If someone sends a PR and there is sufficient interest in this, I will reconsider this.

GGreenix, 2025-01-14:
I am sorry again for opening this thread this would be the last time, I did
add the offsets here
https://github.com/AprilRobotics/apriltag/blob/master/apriltag_quad_thresh.c#L1980
and it caused some issues.

also when you say continue the pipeline you mean all the code also in what
after this line?
https://github.com/AprilRobotics/apriltag/blob/master/apriltag.c#L938
because I saw there are some arithmaticacy being done on the homography in
this line
https://github.com/AprilRobotics/apriltag/blob/master/apriltag.c#L974 as
well
Again sorry for the issues this caused, it's just that I am on a very tight
deadline, thank you for answering.

‫בתאריך יום ד׳, 15 בינו׳ 2025 ב-0:14 מאת ‪Christian Rauch‬‏ <‪
***@***.***‬‏>:‬

s-trinh, 2025-01-15:
> I am sorry again for opening this thread this would be the last time, I did
add the offsets here
https://github.com/AprilRobotics/apriltag/blob/master/apriltag_quad_thresh.c#L1980
and it caused some issues.

You should add the offset in your code, not in the library:

- see the definition of `struct apriltag_detection`: https://github.com/AprilRobotics/apriltag/blob/fc2a7b20a49fc8b211709280e337a76a30db3042/apriltag.h#L196
- see how to retrieve tag info: https://github.com/AprilRobotics/apriltag?tab=readme-ov-file#c
- update the tag 2d image coordinates
- compute the pose with: https://github.com/AprilRobotics/apriltag?tab=readme-ov-file#pose-estimation