AprilRobotics/apriltag #370: Poor tag detecion when tags are rotated.

Frank-Michel, 2025-02-14:
Hi, I am trying to detect tags on a calibration target board (36h11 with double pixel border). Everything is looking okay (first image 24 out of 30 tags detected) when the tags are aligned with the image border. However, when the target is rotated by 45deg (second image) at most 5 tags are getting detected. I was playing with the detector parameters but I was not able to substantially increase the number of detected targets. I was using an altered version of the open_cv example in the repository when running the experiments.

Any advice on why the tags are not detected and how to improve the number of detections is highly appreciated.

[image]

[image]

mkrogius, 2025-02-14:
Unfortunately we do not support the kalibr board. Some previous discussion is [here](https://github.com/AprilRobotics/apriltag/issues/141)

Frank-Michel, 2025-02-18:
Got it. In your linked issue you were talking about the possibility of creating a checkerboard like pattern using Apriltags and inverted Apriltags. I was experimenting with those two types of tags but I was not able to find a solution that does not brake up the solid border around the tags. Could you elaborate on that idea.