AprilRobotics/apriltag #373: AprilTag Preprocessing Steps

Jgunde, 2025-03-14:
This is a general inquiry about preprocessing images before they're fed to the detector.
Are there know ways to improve performance (accuracy, robustness...) by preprocessing images?
Or does the detector already handle any necessary "preprocessing"?

I'm sure there are scenarios and use cases where specific preprocessing steps can be applied to increase performance.
I'm interested in general preprocessing steps like modifying contrast, brightness, saturation, sharpening... that apply to a wide range of scenarios (low-light, distant tags, motion blur, shadows over tags...)

s-trinh, 2025-03-16:
Best things to do in my opinion would be to acquire some data corresponding to your use case and test.

For illumination issues, there are some preprocessing that can be performed easily: histogram equalization , CLAHE algorithm in OpenCV, gamma correction and much more (for instance in [scikit-image](https://scikit-image.org/docs/stable/api/skimage.exposure.html)).

mkrogius, 2025-03-16:
The detector has a number of parameters that can be tuned for your use-case. The `apriltag_demo` binary is the place to start as it has flags that let you modify the detector parameters and the `--debug` flag will cause it to output debug images for each stage of the detector

closed 2025-04-26 by christian-rauch