AprilRobotics/apriltag #374: Advice: how to decode ArUco codeword with AprilTag code?

s-trinh, 2025-03-16:
Hello,

For a separated AprilTag fork, I would like to be able to detect and decode regular ArUco tag (e.g. in OpenCV `DICT_5X5_50`, `DICT_6X6_50`, ...):
- for simplicity between (I believe) the two main fiducial tag options
- the main requirement is to have a white border around the ArUco tag
- to benefit from the AprilTag accurate corners location extraction, (occlusion and detection range are not critical)

So far what I have found/experimented:
- add [`static uint64_t codedata[30]`](https://github.com/AprilRobotics/apriltag/blob/fb2a4096ec5f84b9ec18e501dfa129ffeaaf774e/tag16h5.c#L31) to match ArUco id, for instance in OpenCV for each tag id there are codewords for the 4 rotation ([see](https://github.com/opencv/opencv/blob/6eaaaa410ec63ecb66b48a23dd5396674e2f5125/modules/objdetect/src/aruco/predefined_dictionaries.hpp#L21))
- use `tf->bit_x[0]` and `tf->bit_y[0]` ([see](https://github.com/AprilRobotics/apriltag/blob/fb2a4096ec5f84b9ec18e501dfa129ffeaaf774e/tag16h5.c#L73-L74)) with values that simply correspond to a grid: (x,y) index, left to right, top to bottom: (1,1), (2,1), ...
- add another dedicated `rotate90()` function to produce the 4 codewords for (0, -90°, -180°, -270°) rotations

Ideally, how to:
- generate the `static uint64_t codedata[30]` and the `tf->bit_x[0]`/`tf->bit_y[0]` data for an ArUco family?
- is it possible since I think AprilTag patterns are generated specifically to have some minimal "Hamming distance" between ids to avoid false decoding?
- otherwise would my naive approach (just use (x,y) index for a grid + codeworks corresponding to tag id for one orientation) work without modifying too much code in the detection/decoding pipeline?

My current issues:
- struggle to understand the bits manipulation in `rotate90()`, *when testing it seems to not rotate the codework and probably it has to be related also to the ` tf->bit_x` values*
- struggle to understand the "bucket" and the Hamming, decoding related code, for instance in [`quick_decode_codeword()`](https://github.com/AprilRobotics/apriltag/blob/fb2a4096ec5f84b9ec18e501dfa129ffeaaf774e/apriltag.c#L294), [`quick_decode_init()`](https://github.com/AprilRobotics/apriltag/blob/fb2a4096ec5f84b9ec18e501dfa129ffeaaf774e/apriltag.c#L195), ...

Current quick test:
- I am able to detect and decode an ArUco but only for one rotation with my naive approach (and not sure if the estimated rotation is correct)

Any advice are welcome,
Thanks for the read.

mkrogius, 2025-03-16:
It should be possible to detect ArUcO tags by simply setting `codedata` correctly, without needing to modify `bit_x`/`bit_y` or define a new rotate90 function.

The important thing is to understand the `bit_x`/`bit_y` pattern. Each value in `bit_x`/`bit_y` is a 2d point on the tag at which we will read eventually read out whether a cell is white or black. But these points are not simply row by row (i.e. not a rectangular grid). Instead, imagine you cut an apriltag in half from the top left corner to bottom right corner and then again from the top right corner to bottom left corner. You would then have four triangles. The first quarter of the `bit_x`/`bit_y` values are points on the top triangle, the second quarter are on the right triangle, etc. For the top triangle, the points are read out in rows starting with the top row. And if the size of the tag is odd, then the center location of the tag goes at the end of `bit_x`/`bit_y`.

It's probably simpler with an example. If we had a 3x3 tag, then `bit_x`/`bit_y` would be:
First triangle (x is to the right, y is down):
(1, 1)
(2, 1)

Second triangle:
(3, 1)
(3, 2)

Third triangle:
(3, 3)
(2, 3)

Fourth triangle:
(1, 3)
(1, 2)

Center point:
(2, 2)

I hope that explains the `bit_x`/`bit_y` pattern. If I were to redo things today I would choose a simpler pattern, and you could use a simpler pattern without even changing rotate90 -- the key thing is just that the pattern must have four-fold symmetry. This is necessary so that rotate90 can be implemented efficiently with bitwise operations.

Hopefully understanding the `bit_x`/`bit_y` pattern makes it clear what values should be set in `codedata`. Each bit in a codedata entry is set to indicate whether the tag has a white or black cell at the location given by the `bit_x`/`bit_y` pattern.

s-trinh, 2025-03-19:
Thanks for the explanation.
I think I have grasped the main idea.

---

I am in the process of adding the ArUco family support for another library.
What I have done:
- get the [predefined dictionnary from OpenCV](https://github.com/opencv/opencv/blob/3e43d0cfca9753bcc4983f610b75d70c3f25f0cd/modules/objdetect/src/aruco/predefined_dictionaries.hpp#L4120)
- use the `bit_x`/`bit_y` pattern to get the Hex codeword

For instance for the `DICT_6X6_1000_BYTES`, with my rough code:
- use the following code to get the `static uint64_t codedata` for ArUco
- for the other members, I use corresponding values from the different AprilTag family and only for `tf->h` member (Hamming distance) I use values from the [OpenCV doc](https://docs.opencv.org/4.10.0/de/d67/group__objdetect__aruco.html#gga4e13135a118f497c6172311d601ce00da395a8309569c1977538fd11b57752087)

```c++
void get_36x11_code(unsigned char code1, unsigned char code2, unsigned char code3, unsigned char code4, unsigned char code5)
{
  std::string code_bin = std::bitset<8>(code1).to_string() + std::bitset<8>(code2).to_string()
    + std::bitset<8>(code3).to_string() + std::bitset<8>(code4).to_string() + std::bitset<8>(code5).to_string();
  const int y = 6;

  std::ostringstream oss1, oss2, oss3, oss4, oss5, oss6;
  oss1 << code_bin[0] << code_bin[1] << code_bin[2] << code_bin[3] << code_bin[4] << code_bin[1*y + 1];
  oss2 << code_bin[1*y + 2] << code_bin[1*y + 3] << code_bin[2*y + 2] << code_bin[5] << code_bin[1*y + 5] << code_bin[2*y + 5];
  oss3 << code_bin[3*y + 5] << code_bin[4*y + 5] << code_bin[1*y + 4] << code_bin[2*y + 4] << code_bin[3*y + 4] << code_bin[2*y + 3];
  oss4 << code_bin[5*y + 5] << code_bin[5*y + 4] << code_bin[5*y + 3] << code_bin[5*y + 2] << code_bin[5*y + 1] << code_bin[4*y + 4];
  oss5 << code_bin[4*y + 3] << code_bin[4*y + 2] << code_bin[3*y + 3] << code_bin[5*y + 0] << code_bin[4*y + 0] << code_bin[3*y + 0];
  oss6 << code_bin[2*y + 0] << code_bin[1*y + 0] << code_bin[4*y + 1] << code_bin[3*y + 1] << code_bin[2*y + 1] << code_bin[3*y + 2];

  std::ostringstream oss;
  oss << oss1.str() << oss2.str() << oss3.str() << oss4.str() << oss5.str() << oss6.str();

  std::ostringstream oss_hex;
  oss_hex << std::hex << std::stoll(oss.str(), NULL, 2);
  std::cout << "0x" << std::setfill('0') << std::setw(16) << oss_hex.str() << "UL," << std::endl;
}
```

---

Unfortunately, it does not completely work. On 4x4, for the first tests I have done it seems to work.
On 6x6, certain ID are correctly decoded with some Hamming distance, and the following fails:

[image]

```
decision_margin=0.000000 ; hamming=255
rcode_0:  / 0x00000000069273ca
100000000110100100100111001111001010
rcode_1:  / 0x0000000024e79500
110100100100111001111001010100000000
rcode_2:  / 0x00000000cf2a01a4
100111001111001010100000000110100100
rcode_3:  / 0x0000000054034939
111001010100000000110100100100111001
decision_margin=228.920456 ; hamming=255
```

closed 2025-04-26 by christian-rauch