AprilRobotics/apriltag #375: Tag still detected when slightly occluded

Gold856, 2025-03-31:
**Describe the bug**
When a tag is slightly occluded, such as by being cut off by the top of the image, or by physical objects, the tag is still detected. When doing pose estimation based on these tags, this leads to an incorrect pose due to the fact that some of the detected corners are not actually at the corners of the tag.

**To Reproduce**
Steps to reproduce the behavior:
1. Run opencv_demo.
2. Get an AprilTag and slowly move it up until it goes slightly beyond the camera's field of view.
3. Observe that the tag is still detected and the debug lines move into the tag.

**Expected behavior**
I expected the tag to stop getting detected.

**Screenshots**
[image]

**Operating Sytem**
Windows 11

**Installation Method**
I built AprilTag from source using `cmake -B build -DCMAKE_BUILD_TYPE=Release -G Ninja`.

**Code version**
Current GitHub HEAD.

christian-rauch, 2025-03-31:
Can you show the output of the `apriltag_demo` on this example image? Maybe you can filter this detection based on the decision margin.

christian-rauch, 2025-03-31:
That's quite a large margin. Just compare with good "true" detections and select an appropriate threshold.

Gold856, 2025-03-31:
I seem to have gotten a much lower margin with this image:
[image]

```
detection   0: id (36x11)-1   , hamming 0, margin  149.803
 0                             init        0.000000 ms        0.000000 ms
 1                         decimate        0.000000 ms        0.000000 ms
 2                       blur/sharp        0.000000 ms        0.000000 ms
 3                        threshold        0.000000 ms        0.000000 ms
 4                        unionfind        0.000000 ms        0.000000 ms
 5                    make clusters        0.000000 ms        0.000000 ms
 6            fit quads to clusters        0.000000 ms        0.000000 ms
 7                            quads        0.000000 ms        0.000000 ms
 8                decode+refinement        0.000000 ms        0.000000 ms
 9                        reconcile        0.000000 ms        0.000000 ms
10                     debug output        0.000000 ms        0.000000 ms
11                          cleanup        0.000000 ms        0.000000 ms
hamm     1     0     0     0     0     0     0     0     0     0        0.000     7
Summary
hamm     1     0     0     0     0     0     0     0     0     0        0.000     7
```

```

christian-rauch, 2025-03-31:
> So when I'm doing pose estimation, I want to reject detections with a high margin to keep out cut off tags right?

Yes. That was my initial suggestion.

Also, have you tried other families? You can generate your own tag families with large hamming distance, if the defaults are not robust enough.

Gold856, 2025-03-31:
We have not tried other families. Our robotics competition [seems to be sticking with 36h11 tags after having tried 16h5 tags](https://www.firstinspires.org/robotics/frc/blog/2023-technology-updates-past-present-future-and-beyond), and they are robust except for [edge cases regarding tags being cut off](https://github.com/PhotonVision/photonvision/issues/1817). So we were wondering how we could fix that, and it looks like we can solve this by filtering out high margin tag detections. Thank you for help!

closed 2025-03-31 by Gold856

Gold856, 2025-03-31:
Actually, I want clarify something about decision margin. You said I should filter out tags with a high decision margin, but [apriltag.h](https://github.com/AprilRobotics/apriltag/blob/master/apriltag.h#L212) says that higher numbers actually indicate a better tag decode? I'm a bit confused now on why I should filter out detections with a high decision margin.

christian-rauch, 2025-03-31:
You should reject detections with a low margin, see https://github.com/AprilRobotics/apriltag/issues/361 for an open discussion on this. There, the difference between the true detection and the false positive is very distinct.

mcm001, 2025-03-31:
Heyo! Just chiming in again (I work on the same project as Gold856). We already reject (with great success, in particular for 16h5) tags with a sufficiently low decision margin. For our testing, that's ~30. In this issue, it looks like the example images Gold posted have a decision margin on the order of 150, with one of the cut-off tags being at 180?

I also noticed [these lines](https://github.com/AprilRobotics/apriltag/blob/fb2a4096ec5f84b9ec18e501dfa129ffeaaf774e/apriltag.h#L210C1-L216C33) in apriltag.h about decision margin not being applicable for large tags -- is this relevant?

```
    // A measure of the quality of the binary decoding process: the
    // average difference between the intensity of a data bit versus
    // the decision threshold. Higher numbers roughly indicate better
    // decodes. This is a reasonable measure of detection accuracy
    // only for very small tags-- not effective for larger tags (where
    // we could have sampled anywhere within a bit cell and still
    // gotten a good detection.)
    float decision_margin;
```

At its core, I think my ask is, "for sufficiently large tag, what can I do to make sure that there's a white boarder around my tag"?

mkrogius, 2025-04-03:
If the concern is only about the tag getting cut off by the edge of the image (and not about tags getting occluded by obstacles), then you can check that the corner locations are sufficiently far from the image borders. The other solution would be to switch tag families to the tagStandard41h12 family, which would not be detected in these scenarios because the layout places data bits on the exterior

mcm001, 2025-04-08:
Do you have a good heuristic for what "sufficiently far" might mean? We can experiment/do some science ourselves of course. Just not sure what the most robust way to try that would be, since it looks like in this example, the corner point doesn't lie on the edge of the image -- but it is quite close.

[image]

mkrogius, 2025-04-10:
The other thing I should mention is that you can adjust the max_line_fit_mse param to filter out cases like this