AprilRobotics/apriltag #378: Why there is a memory leakage in detection

WZG3661, 2025-04-22:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
[e.g. Run apriltag_demo on the attached input image.]

**Expected behavior**
A clear and concise description of what you expected to happen.

** Input Image**
If applicable, please attach the input image that reproduces the problem.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Operating Sytem**
[e.g. Ubuntu, iOS, Windows]

** Installation Method**
[e.g. I built AprilTag from source following the instructions in the README]

** Code version **
What does the version tag in package.xml say? Or, if you have downloaded the code from github right now, say current github HEAD.

**Additional context**
Add any other context about the problem here.

christian-rauch, 2025-04-22:
Can you be more precise what the problem is?

WZG3661, 2025-04-22:
I found if I don't paralise the left and right detection, the memory usage not increase:

[image]

christian-rauch, 2025-04-22:
Just consuming more memory does not automatically mean a memory leak. Obviously, the memory consumptions varies with the number of detections you have to store, which likely varies from image to image.

A memory leak is when you allocate memory and do not deallocate / free the memory again. Can you show with a tool like valgrind or AddressSanitizer, or in the code, that there is an actual memory leak?

WZG3661, 2025-04-22:
but I didn't store any result of detections, once I finished detection, I destroy it.

christian-rauch, 2025-04-22:
The apriltag library stores detections in an array. The more detections, the larger the array.

WZG3661, 2025-04-22:
Yes, but I destroy it after detection:

[image]
Why the memory usage still increase

christian-rauch, 2025-04-22:
Still, if later images have more detections than earlier images, the library will allocate more memory for the detections.

Can you provide an example where you loop over the very same image, i.e. the detections will always be the same, and show that the memory increases over time? You may want to use the `apriltag_demo.c` as a starting point. Share your code and the example image to reproduce this.

christian-rauch, 2025-04-22:
> The code is above and the image is not same, but they don't have any tags, they are just common images from a video

Yes, I know that your code above does not do what I asked. But I am asking you to provide said example for a single image in order to show that there is a memory leak.

closed 2025-04-26 by christian-rauch