AprilRobotics/apriltag #379: Proposal for OpenHarmony Adaptation of [ apriltag ]

techvision0628, 2025-05-09:
## Background
OpenHarmony is an open-source operating system and community that is widely used in the field of intelligent terminals.

Apriltag is a commonly used visual fiducial system in robotics research. In order to enable apriltag to support the OpenHarmony system, we have modified some of the code of apriltag. We plan to submit the modified code to the apriltag community. The purpose of creating this Issue is to initiate discussions with the community. We sincerely welcome you to put forward your valuable suggestions.  

## Adaptation Proposal
We have completed the adaptation of apriltag on OpenHarmony. The specific content is as follows:

- The construction/packaging of apriltag was realized based on the OpenHarmony environment and development tools.
- Provide sample examples of using apriltag in OpenHarmony application development.

We'll submit the adaptation code through a Pull Request (PR) promptly.

## Testing Results
Based on the OpenHarmony system, we conducted unit tests on apriltag using the original library test cases and passed the tests in the following versions:

DevEco Studio 5.0.3 Release-5.0.9.221, SDK: API15(OpenHarmony 5.0.3.135)  
DevEco Studio 5.0.4 Beta1-5.0.9.320, SDK: API16 (OpenHarmony 5.0.4.150)  
DevEco Studio 5.0.3 Release-5.0.9.300, SDK: API15(OH_SDK 5.0.3.135(SP1))  
DevEco Studio 5.0.4 Beta1 5.0.9.320, SDK: API16(OH SDK 5.0.4.150)

If necessary, we can provide detailed test reports at any time.

## Labels
enhancement, compatibility, cross-platform

christian-rauch, 2025-05-09:
I would wait for the promised PR to see the details.

According to Wikipedia, [OpenHarmony](https://en.wikipedia.org/wiki/OpenHarmony) is POSIX compatible. Since the library does not have any dependencies, except the optional dependency on OpenCV, I don't expect many changes apart from adjustments for alternative compilers.

plusvic, 2025-05-12:
This looks like spam to me.l The same message has been posted in other repositories as well: https://github.com/VirusTotal/yara/issues/2151

techvision0628, 2025-05-13:
> This looks like spam to me. The same message has been posted in other repositories as well: [VirusTotal/yara#2151](https://github.com/VirusTotal/yara/issues/2151)

Not spam, we adapted some third-party libraries，so we wrote the same communication format for multiple different third-party libraries.

techvision0628, 2025-05-20:
> [@techvision0628](https://github.com/techvision0628) Can you elaborate when
> 
> > We'll submit the adaptation code through a Pull Request (PR) promptly.
> 
> will happen? Are you planning on submitting a PR, or can I close this issue?

We are currently organizing the PRs to be submitted.
Since the PRs to be submitted also need to be reviewed by several other people, I cannot give you accurate time.
This issue can be closed at the moment, it does not affect the subsequent submission of PRs.

gsdukbh, 2025-05-20:
This user is an automated bot from Huawei. They are creating issues in various repositories to advertise their products.

christian-rauch, 2025-06-15:
It looks like this is never going to happen and all these OpenHarmony PRs are just vaporware to advertise OpenHarmony. I am closing this. Just send the PRs where there is something to fix. But please stop opening any such PRs to any of the repos I maintain.