AprilRobotics/apriltag #380: Any examples of annular AprilTags?

sgbaird, 2025-05-15:
Interested for physical science applications (e.g., beakers, vials) to help with locating grasp positions.

The possibility for annular tags was mentioned in https://april.eecs.umich.edu/papers/details.php?name=krogius2019iros, but I've had trouble finding an example somewhere. Any advice or a starting point would be welcome

sgbaird, 2025-05-15:
Thanks! So I guess the idea of wrapping an AprilTag on a curved surface isn't really supported/demonstrated. Thanks for the link to the flexible layout images

mkrogius, 2025-05-15:
Correct, it's not demonstrated or supported. But it will still work to a degree. The detector will detect the tag as long as the deformation doesn't move any part of the image of the tag by more than half of a cell, which is a metric that depends on both the amount of curvature and also the viewpoint. If the tag is successfully detected, pose estimation will work normally as long as you supply the ground truth corner positions of the curved tag. The apriltag library doesn't support this but you could use OpenCV PNP to do this

closed 2025-06-15 by christian-rauch